JuliaControl/ModelPredictiveControl.jl #26: Dev documentation no longer deployed

franckgaga, 2024-02-09:
@baggepinnen 

It seems that dev documentation is no longer automatically built and deployed when I commit on `main`. It was working perfectly before transferring to JuliaControl. Do you have any idea why ? It may be related to secrets key (in documentation.yml):
```yml
env:
          GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }} # If authenticating with GitHub Actions token
          DOCUMENTER_KEY: ${{ secrets.DOCUMENTER_KEY }} # If authenticating with SSH deploy key
```
I'm no longer able to access the secrets in repo settings since I'm not an admin.

Also, I just registered a new version. I'm waiting to see if the stable doc will be built and deployed.

baggepinnen, 2024-02-10:
I'll have a look when I reach a computer 😊

baggepinnen, 2024-02-12:
Closed by https://github.com/JuliaControl/ModelPredictiveControl.jl/commit/90168108e956675add753b2018f067324259e1d0

closed 2024-02-12 by baggepinnen

franckgaga, 2024-02-12:
Lol that was way easier than I expected, thanks! :)