JuliaControl/ModelPredictiveControl.jl #76: MPC getinfo results

caxelrud, 2024-06-06:
Hi, I am getting the following results when using getinfo().
N is 200.
it looks like, for multivariable inputs and outputs, the results are flattened but clamped to the size N.
```
info=getinfo(mpc)
info[:Rhaty ]
julia>
200-element Vector{Float64}:
 50.0
 30.0
  ⋮
 50.0
 30.0
```

franckgaga, 2024-06-06:
I'm not able to reproduce, are you using the last version v.0.21.3 ? Did you did you solve the mpc problem at least one time with `moveinput!` ?

Here's the result on my side:

```julia
using ModelPredictiveControl, ControlSystemsBase
mpc = LinMPC(LinModel(append(tf(2,[10,1]), tf(3,[5, 1])), 1.0), Hp=200);
u = moveinput!([2,10]);
info = getinfo!(mpc);
info[:Rhaty]
```
results in:
```
400-element Vector{Float64}:
  2.0
 10.0
  2.0
 10.0
  2.0
  ⋮
  2.0
 10.0
  2.0
 10.0
```

caxelrud, 2024-06-07:
I found a bug in my testing code.
All looks good now so, please, disregard my issue.

franckgaga, 2024-06-07:
Alright, let me know if there is another issue !

closed 2024-06-07 by franckgaga