JuliaControl/ModelPredictiveControl.jl #98: Support Joseph-form in Kalman filters

franckgaga, 2024-09-11:
It would be nice to support the Joseph-form covariance update equation in `KalmanFilter`, `ExtendedKalmanFilter` and `UnscentedKalmanFilter`. It theoretically guarantee positive-definite covariance matrices, even with finite precision floating-point arithmetic. 

In my experience although, I never had any problems with the classical equation, both with the Kalman filter and the unscented version. I personally don't consider that as a high priority feature.

franckgaga, 2025-01-22:
I noticed that once and a while in the `MovingHorizonEstimator` tests, the estimation fails because of the arrival covariance $\mathbf{\bar{P}}$ estimation based on `UnscentedKalmanFilter`, that internally uses `cholesky`. The error is `PosDefException: matrix is not positive definite; Factorization failed.`

It raises two points:
1. Should I encapsulate arrival covariance estimation of the MHE in a `try` block ? If yes what would be the fallback ? Do not update the arrival covariance (keep the one from the last iteration $k-1$) and emit a `@warn` ? I think It would be a good idea since with a reasonably long `He`, the impact of the arrival estimate term in the cost function is typically quite low. Morever, I frequently used the MHE with a constant arrival covariance estimate in the past and it was simpler and working quite well.
2. The Joseph-From should be implemented in the package and be the default for the `MovingHorizonEstimator` arrival covariance estimation. The "numerical noise" produced by the MHE optimization is more risky to produce negative semidefinite covariance matrix.

Do you have an opinion @baggepinnen ?

Thanks!

baggepinnen, 2025-01-23:
I include `P̄ = 0.5*(P̄ + P̄')` in more or less all code that modifies covariance matrices, I find that this is often all that is required to handle numerical issues in Kalman filters

franckgaga, 2025-01-23:
I'll also add this line as another "saftey net" for the MHE.

1-Bart-1, 2025-03-25:
> I noticed that once and a while in the `MovingHorizonEstimator` tests, the estimation fails because of the arrival covariance P ¯ estimation based on `UnscentedKalmanFilter`, that internally uses `cholesky`. The error is `PosDefException: matrix is not positive definite; Factorization failed.`

I have gotten this error several times as well.

franckgaga, 2025-03-25:
@1-Bart-1 If you a MWE that systematically produce this error, send it to me. I will try to investigate on the error. Fredrik and I emit the hypothesis that It may be caused by the default values of the UKF $\alpha, \beta, \kappa$ weights (the defaults are not rigorously justified, it's just values that many people use without a real justification). 

If that's really the case, a breaking change may be worthwhile.

baggepinnen, 2025-03-25:
@1-Bart-1 see this comment and the linked article for more details on the UT parameters
https://github.com/baggepinnen/LowLevelParticleFilters.jl/pull/191#issuecomment-2712620825

1-Bart-1, 2025-03-25:
I don't have a MWE... If I remember correctly, the error tends to disappear when changing some of the estimator weights. But I will definitely come back here if the error appears again.

1-Bart-1, 2025-04-17:
I am getting this issue again. I don't exactly have a lightweight example, but you can run `mwe/online_lin_simple_kite.jl` to recreate the issue from this branch: [KitePredictiveControl.jl#cholesky](https://github.com/Albatross-Kite-Transport/KitePredictiveControl.jl).

1-Bart-1, 2025-04-17:
I am getting this issue again. I don't have a lightweight example, but you can recreate the issue by running [mwe/online_lin_simple_kite.jl](https://github.com/Albatross-Kite-Transport/KitePredictiveControl.jl/blob/cholesky/mwe/online_lin_simple_kite.jl) from [KitePredictiveControl.jl#cholesky](https://github.com/Albatross-Kite-Transport/KitePredictiveControl.jl/tree/cholesky). Let me know if you have any issues recreating this. Is it possible to play around with the weights in MPC.jl currently?

franckgaga, 2025-04-17:
Thanks for the exemple! Above I was mentioning an issue with the MHE that uses the UKF internally for the arrival covariance estimation. So it's not related to the UKF $\alpha, \beta and \kappa$ weights.

Cholesky is used in `KalmanFilter` for array division (whitout any allocation, `/` and 
`\` operators are allocating). It's kind of a similar as the issue above: the covariance estimate $\mathbf{P}$ are no longer positive definite at some points in the simulation, probably because or numerical issues. I'll take a look at you exemple. It's possible that Joseph-form would be a solution.