JuliaControl/ModelPredictiveControl.jl #109: Support for dynamic differential programming in a model predictive context

acxz, 2024-09-30:
Congrats on the 1.0.0 release!

I wanted to ask if this library supports [DDP](https://en.wikipedia.org/wiki/Differential_dynamic_programming) in a MPC manner? And if not, is support planned for this? Maybe this support would be seen as a composition of this library and an optimal control library? Or this optimization feature expected to come from JuMP.jl directly?

Some DDP context in the Julia community:
For a DDP backend there is [this](https://github.com/baggepinnen/DifferentialDynamicProgramming.jl) library, but it is effectively unmaintained and archived. The author @baggepinnen has pointed me to RoboticExplorationLab's [Altro.jl](https://github.com/RoboticExplorationLab/ALTRO.jl).

baggepinnen, 2024-09-30:
DDP is a very different paradigm to what is employed in this package, adding it would be a very nontrivial task and is probably out of scope for this package, it's certainly not as simple as "adding a DDP backend".

If you want to use DDP, why not use Altro.jl directly?

acxz, 2024-10-01:
I don't just want to use DDP, but there are other optimal controllers like MPPI that I'd like to implement. It makes sense for me to use some common interface. There is TrajectoryOptimization.jl, but that is outside the purview of JuliaControl. I guess I'm looking for a JuliaControl project that supports problems that control algs like DDP solve. I thought that would be this package with the use of JuMP for the actual optimization.

baggepinnen, 2024-10-01:
>  I guess I'm looking for a JuliaControl project that supports problems that control algs like DDP solve

There is none. Why does it have to be in JuliaControl? 

> I thought that would be this package with the use of JuMP for the actual optimization.

I don't think there are any DDP solvers that support MathOptInterface that JuMP uses, probably because JuMP by itself does not provide you with the structure you need for DDP-style problem formulation.

franckgaga, 2024-10-02:
I am not familiar with DDP. Although there is some similarities DDP and MPC from my quick overview, the underlying frameworks seem quite different. DDP seems to be more of an open-loop control policy ? This org. is restricted to closed-loop control.

From my perspective, there is way too much different MPC policies to potentially add to this package (multiple shooting MPC, direct collocation, robust MPC, etc.) before considering adding non-MPC approaches.

acxz, 2024-10-03:
> Why does it have to be in JuliaControl?

I guess the reason that I thought having an implementation be in JuliaControl would be that there is some type that my controller can be so that I can compose with other functionality that JuliaControl may offer.

> I don't think there are any DDP solvers that support MathOptInterface that JuMP uses, probably because JuMP by itself does not provide you with the structure you need for DDP-style problem formulation.

I see, I'm still getting familiar with the ecosystem here. A quick search results in a JuMP extension that does have support for dynamic programming: https://jump.dev/JuMP.jl/stable/packages/InfiniteOpt/.

> Although there is some similarities DDP and MPC from my quick overview, the underlying frameworks seem quite different.

It doesn't make sense to compare DDP and MPC. From what I understand of MPC (or receding horizon control), it is how you apply a control policy. DDP can be applied in an MPC fashion or it cannot. You can formulate your trajopt/control problem as an SQP problem and apply that in an MPC fashion (Or not), etc.

Just as further reading ["A Family of Iterative Gauss-Newton Shooting Methods for Nonlinear
Optimal Control"](https://arxiv.org/pdf/1711.11006) is a paper that generalizes DDP (the single derivative approx version, also called iLQR) in a multiple shooting context. Maybe this helps in trying to explain what I'm talking about.

I apologize for the noise, but I did find this conversation useful.

baggepinnen, 2024-10-03:
InfiniteOpt does not use DDP, it uses a collocation on finite elements which is solved by a standard NLP solver like Ipopt.

franckgaga, 2024-10-04:
I will close the issue but fill free to continue the discussion or re-open it if needed!

closed 2024-10-04 by franckgaga