JuliaControl/ModelPredictiveControl.jl #119: plot not working inside module

1-Bart-1, 2024-10-17:
I am using MPC.jl in a module. When running `ModelPredictiveControl.plot(res; plotx=true, ploty=true, plotu=true, plotxwithx̂=true)` I get the error: 
```julia
ERROR: MethodError: no method matching plot(::ModelPredictiveControl.SimResult{Float64, ModelPredictiveControl.LinMPC{…}}; plotx::Bool, ploty::Bool, plotu::Bool, plotxwithx̂::Bool)

Closest candidates are:
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.PredictiveController}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:325
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.StateEstimator}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:324
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.SimModel}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:323

Stacktrace:
 [1] plot(ci::ControlInterface)
   @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/KitePredictiveControl.jl:209
 [2] top-level scope
   @ ./REPL[17]:5
Some type information was truncated. Use `show(err)` to see complete types.
```

I don't know what could cause this. It worked before, but after some code reformatting, it stopped working.

```julia
julia> typeof(res)
SimResult{Float64, LinMPC{Float64, KalmanFilter{Float64, LinModel{Float64}}, JuMP.Model}}

julia> res.obj
LinMPC controller with a sample time Ts = 0.05 s, DAQP optimizer, KalmanFilter estimator and:
 20 prediction steps Hp
  4 control steps Hc
  1 slack variable ϵ (control constraints)
  3 manipulated inputs u (3 integrating states)
 78 estimated states x̂
 26 measured outputs ym (23 integrating states)
  0 unmeasured outputs yu
  0 measured disturbances d
  
  julia> res.X_data
52×800 Matrix{Float64}:
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  …  0.0  0.0  0.0  0.0  0.0  0.0  12.1372       12.4948    12.1372       12.1184
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   0.420671      0.398937   0.420671      0.420346
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  26.1225       24.0023    26.1225       26.1313
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  12.1372       12.4948    12.1372       12.1184
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  -0.420671     -0.398937  -0.420671     -0.420346
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  …  0.0  0.0  0.0  0.0  0.0  0.0  26.1225       24.0023    26.1225       26.1313
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   2.23948       2.17889    2.23948       2.23148
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   2.14705e-11   0.0        1.76796e-11   1.65288e-11
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  25.0963       24.9049    25.0963       25.102
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   0.14923       0.0        0.14923       0.15375
 ⋮                        ⋮                        ⋮                        ⋮              ⋱  ⋮                        ⋮                                           
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   0.308326      0.0        0.308326      0.210995
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  -0.338401      0.0       -0.338401     -0.269748
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   5.05014e-12   0.0       -7.45084e-12  -4.56091e-11
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  …  0.0  0.0  0.0  0.0  0.0  0.0   0.307788      0.0        0.307788      0.209391
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  57.611        57.6066    57.611        57.611
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  57.611        57.6066    57.611        57.611
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0  50.1319       50.0       50.1319       50.1404
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   0.00112142    0.0        0.00112142    0.0011503
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  …  0.0  0.0  0.0  0.0  0.0  0.0   0.00112142    0.0        0.00112142    0.0011503
 0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0  0.0     0.0  0.0  0.0  0.0  0.0  0.0   0.16879       0.0        0.16879       0.169753
```

baggepinnen, 2024-10-17:
I would guess that you do something wrong in your package, how is this method defined? is it using RecipesBase.jl or directly overloading `Plots.plot`?
```julia
plot(ci::ControlInterface)
   @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/KitePredictiveControl.jl:209
```

1-Bart-1, 2024-10-17:
It could very well be that I am doing something wrong in my package. But changing the package `plot` function to `controlplot` does not solve the issue:

```julia
ERROR: LoadError: MethodError: no method matching plot(::ModelPredictiveControl.SimResult{Float64, ModelPredictiveControl.LinMPC{…}}; plotx::Bool, ploty::Bool, plotu::Bool, plotxwithx̂::Bool)

Closest candidates are:
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.PredictiveController}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:325
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.StateEstimator}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:324
  plot(::Nothing, ::ModelPredictiveControl.SimResult{<:Real, <:ModelPredictiveControl.SimModel}) got unsupported keyword arguments "plotx", "ploty", "plotu", "plotxwithx̂"
   @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/plot_sim.jl:323

Stacktrace:
 [1] controlplot(ci::ControlInterface)
   @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/KitePredictiveControl.jl:210
 [2] top-level scope
   @ ~/Code/KitePredictiveControl.jl/examples/simple.jl:16
 [3] include(fname::String)
   @ Base.MainInclude ./client.jl:489
 [4] top-level scope
   @ REPL[1]:1
in expression starting at /home/bart/Code/KitePredictiveControl.jl/examples/simple.jl:12
Some type information was truncated. Use `show(err)` to see complete types.
```

The method is defined in `module KitePredictiveControl` as:

```julia
function controlplot(ci::ControlInterface)
    res = SimResult(ci.mpc, ci.U_data, ci.Y_data; ci.Ry_data, ci.X̂_data, ci.X_data)
    # return ModelPredictiveControl.plot(res; plotx=true, ploty=ci.output_idxs, plotu=true, plotxwithx̂=ci.observed_idxs)

    return ModelPredictiveControl.plot(res; plotx=true, ploty=true, plotu=true, plotxwithx̂=true)
end
```

baggepinnen, 2024-10-17:
It looks like this is the culprit
https://github.com/JuliaControl/ModelPredictiveControl.jl/blob/38333015ea653e601ed1955bed2f1142dddc64a7/src/plot_sim.jl#L323

As a fix for you, I would try calling `Plots.plot` instead of `ModelPredictiveControl.plot`

1-Bart-1, 2024-10-17:
Thanks, that solved the issue. Maybe this should be documented to avoid future confusions?

franckgaga, 2024-11-21:
Closed by #120, thanks for the report! :)

closed 2024-11-21 by franckgaga