JuliaControl/ModelPredictiveControl.jl #121: linearize is throwing an error when using an empty disturbance

1-Bart-1, 2024-10-22:
I don't have disturbances implemented in my model.
```julia
julia> @show nonlinmodel.buffer.d .- nonlinmodel.dop
nonlinmodel.buffer.d .- nonlinmodel.dop = Float64[]
```
The linearization shouldn't need to run on the disturbances, but it does, and this causes an error with the following error/stacktrace:
```julia
ERROR: LoadError: MethodError: reducing over an empty collection is not allowed; consider supplying `init` to the reducer
Stacktrace:
  [1] reduce_empty(op::Base.MappingRF{typeof(DiffEqBase.sse), typeof(Base.add_sum)}, ::Type{Float64})
    @ Base ./reduce.jl:361
  [2] reduce_empty_iter
    @ ./reduce.jl:384 [inlined]
  [3] mapreduce_empty_iter(f::Function, op::Function, itr::ForwardDiff.Partials{0, Float64}, ItrEltype::Base.HasEltype)
    @ Base ./reduce.jl:380
  [4] _mapreduce
    @ ./reduce.jl:432 [inlined]
  [5] _mapreduce_dim
    @ ./reducedim.jl:365 [inlined]
  [6] mapreduce
    @ ./reducedim.jl:357 [inlined]
  [7] _sum
    @ ./reducedim.jl:1015 [inlined]
  [8] sum(f::Function, a::ForwardDiff.Partials{0, Float64})
    @ Base ./reducedim.jl:1011
  [9] sse(x::ForwardDiff.Dual{ForwardDiff.Tag{ModelPredictiveControl.var"#myf_d0!#37"{…}, Float64}, Float64, 0})
    @ DiffEqBase ~/.julia/packages/DiffEqBase/frOsk/src/forwarddiff.jl:452
 [10] MappingRF
    @ ./reduce.jl:100 [inlined]
 [11] _foldl_impl
    @ ./reduce.jl:58 [inlined]
 [12] foldl_impl
    @ ./reduce.jl:48 [inlined]
 [13] mapfoldl_impl
    @ ./reduce.jl:44 [inlined]
 [14] _mapreduce_dim
    @ ./reducedim.jl:362 [inlined]
 [15] mapreduce
    @ ./reducedim.jl:357 [inlined]
 [16] _sum
    @ ./reducedim.jl:1015 [inlined]
 [17] sum
    @ ./reducedim.jl:1011 [inlined]
 [18] __sum
    @ ~/.julia/packages/DiffEqBase/frOsk/src/forwarddiff.jl:483 [inlined]
 [19] ODE_DEFAULT_NORM
    @ ~/.julia/packages/DiffEqBase/frOsk/src/forwarddiff.jl:462 [inlined]
 [20] __init(prob::SciMLBase.ODEProblem{…}, alg::OrdinaryDiffEqBDF.QNDF{…}, timeseries_init::Tuple{}, ts_init::Tuple{}, ks_init::Tuple{}, recompile::Type{…}; saveat::Float64, tstops::Tuple{}, d_discontinuities::Tuple{}, save_idxs::Nothing, save_everystep::Bool, save_on::Bool, save_start::Bool, save_end::Nothing, callback::Nothing, dense::Bool, calck::Bool, dt::Float64, dtmin::Float64, dtmax::Float64, force_dtmin::Bool, adaptive::Bool, gamma::Rational{…}, abstol::Nothing, reltol::Nothing, qmin::Rational{…}, qmax::Int64, qsteady_min::Int64, qsteady_max::Rational{…}, beta1::Nothing, beta2::Nothing, qoldinit::Rational{…}, controller::Nothing, fullnormalize::Bool, failfactor::Int64, maxiters::Int64, internalnorm::typeof(DiffEqBase.ODE_DEFAULT_NORM), internalopnorm::typeof(LinearAlgebra.opnorm), isoutofdomain::typeof(DiffEqBase.ODE_DEFAULT_ISOUTOFDOMAIN), unstable_check::typeof(DiffEqBase.ODE_DEFAULT_UNSTABLE_CHECK), verbose::Bool, timeseries_errors::Bool, dense_errors::Bool, advance_to_tstop::Bool, stop_at_next_tstop::Bool, initialize_save::Bool, progress::Bool, progress_steps::Int64, progress_name::String, progress_message::typeof(DiffEqBase.ODE_DEFAULT_PROG_MESSAGE), progress_id::Symbol, userdata::Nothing, allow_extrapolation::Bool, initialize_integrator::Bool, alias_u0::Bool, alias_du0::Bool, initializealg::OrdinaryDiffEqCore.DefaultInit, kwargs::@Kwargs{})
    @ OrdinaryDiffEqCore ~/.julia/packages/OrdinaryDiffEqCore/NnA60/src/solve.jl:329
 [21] __init (repeats 5 times)
    @ ~/.julia/packages/OrdinaryDiffEqCore/NnA60/src/solve.jl:11 [inlined]
 [22] __solve(::SciMLBase.ODEProblem{…}, ::OrdinaryDiffEqBDF.QNDF{…}; kwargs::@Kwargs{…})
    @ OrdinaryDiffEqCore ~/.julia/packages/OrdinaryDiffEqCore/NnA60/src/solve.jl:6
 [23] __solve
    @ ~/.julia/packages/OrdinaryDiffEqCore/NnA60/src/solve.jl:1 [inlined]
 [24] #solve_call#44
    @ ~/.julia/packages/DiffEqBase/frOsk/src/solve.jl:612 [inlined]
 [25] solve_call
    @ ~/.julia/packages/DiffEqBase/frOsk/src/solve.jl:569 [inlined]
 [26] #solve_up#53
    @ ~/.julia/packages/DiffEqBase/frOsk/src/solve.jl:1092 [inlined]
 [27] solve_up
    @ ~/.julia/packages/DiffEqBase/frOsk/src/solve.jl:1078 [inlined]
 [28] #solve#51
    @ ~/.julia/packages/DiffEqBase/frOsk/src/solve.jl:1015 [inlined]
 [29] (::KitePredictiveControl.var"#next_step!#22"{…})(x_plus::Base.ReinterpretArray{…}, x::Vector{…}, u::Vector{…}, prob::SciMLBase.ODEProblem{…})
    @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/mtk_interface.jl:20
 [30] (::KitePredictiveControl.var"#f!#23"{…})(x_plus::Base.ReinterpretArray{…}, x::Vector{…}, u::Vector{…}, ::Vector{…}, ::Vector{…})
    @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/mtk_interface.jl:8
 [31] (::ModelPredictiveControl.var"#inner_solver_f!#19"{…})(xnext::Vector{…}, x::Vector{…}, u::Vector{…}, d::Vector{…}, p::Vector{…})
    @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/solver.jl:61
 [32] f!
    @ ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/nonlinmodel.jl:208 [inlined]
 [33] myf_d0!
    @ ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/linearization.jl:126 [inlined]
 [34] vector_mode_dual_eval!
    @ ~/.julia/packages/ForwardDiff/PcZ48/src/apiutils.jl:31 [inlined]
 [35] vector_mode_jacobian!(result::Matrix{…}, f!::ModelPredictiveControl.var"#myf_d0!#37"{…}, y::Vector{…}, x::Vector{…}, cfg::ForwardDiff.JacobianConfig{…})
    @ ForwardDiff ~/.julia/packages/ForwardDiff/PcZ48/src/jacobian.jl:153
 [36] jacobian!(result::Matrix{…}, f!::Function, y::Vector{…}, x::Vector{…}, cfg::ForwardDiff.JacobianConfig{…}, ::Val{…})
    @ ForwardDiff ~/.julia/packages/ForwardDiff/PcZ48/src/jacobian.jl:78
 [37] jacobian!(result::Matrix{…}, f!::Function, y::Vector{…}, x::Vector{…}, cfg::ForwardDiff.JacobianConfig{…})
    @ ForwardDiff ~/.julia/packages/ForwardDiff/PcZ48/src/jacobian.jl:76
 [38] linearize!(linmodel::ModelPredictiveControl.LinModel{…}, model::ModelPredictiveControl.NonLinModel{…}; x::Vector{…}, u::Vector{…}, d::Vector{…})
    @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/linearization.jl:131
 [39] linearize!
    @ ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/linearization.jl:110 [inlined]
 [40] linearize(model::ModelPredictiveControl.NonLinModel{…}; kwargs::@Kwargs{…})
    @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/0rhFm/src/model/linearization.jl:87
 [41] macro expansion
    @ ./timing.jl:279 [inlined]
 [42] ControlInterface(kite::KPS4_3L{…}; Ts::Float64, u0::Vector{…})
    @ KitePredictiveControl ~/Code/KitePredictiveControl.jl/src/KitePredictiveControl.jl:78
 [43] top-level scope
    @ ~/Code/KitePredictiveControl.jl/examples/simple.jl:15
 [44] include(fname::String)
    @ Base.MainInclude ./client.jl:489
 [45] top-level scope
    @ REPL[2]:1
in expression starting at /home/bart/Code/KitePredictiveControl.jl/examples/simple.jl:15
Some type information was truncated. Use `show(err)` to see complete types.
```

It would be nice if this line could be hopped over if there are no disturbances implemented:
`    myf_d0!(xnext0, d0) = f!(xnext0, nonlinmodel, x0, u0, d0, model.p)`

franckgaga, 2024-10-23:
What's the code that produces the error above ?

Do you have a MWE ?

1-Bart-1, 2024-10-23:
Thanks, I forgot to set solver to nothing... This solved the error.

closed 2024-10-23 by 1-Bart-1