JuliaControl/ModelPredictiveControl.jl #124: Code generation

ufechner7, 2024-11-11:
What would be the way forward to become able to run an MPC controller on a small microcontroller?

Other packages implement C code generation. Would that be possible, or what could be alternatives?

baggepinnen, 2024-11-12:
There is no code generation from this package yet, and it's not a small task to implement. One possible way forward is to use the code-generation capabilities of an existing MPC package with this feature, such as acados. I have a proof-of-concept doing that here https://github.com/baggepinnen/AcadosInterface.jl You would have to manually create the acados problem by passing the same cost matrices to acados as you do to MPC.jl, but this should be quite straightforward. It's limited to things that are easy to express in both Symbolics.jl and CasADi, so it does not accept any arbitrary dynamics.

franckgaga, 2024-11-12:
One issue with code generation in general is how it can be hard to exactly reproduce results. I expect to see difference between MPC.jl results and the Acados interface (different solvers, state estimators, linear algebra libraries, etc.). Hopefully, these differences will be negligible for well-conditioned problems.

In the end code generation is just another form of the two-language problem. It may be kind of a long shot prediction, but I think the future of MPC lies in more advanced control hardware, like a full-fledge computer or server instead of embedded platform. Micro-controllers and PLCs computational capabilities are more than enough for PIDs, but the layers of complexity in MPC (solvers, automatic differentiation libraries, linear algebra libraries, etc.) clearly benefits from Julia's modern features like the composability, the package manager, etc.

Still, code generation features for this package would be extremely useful right now!

baggepinnen, 2025-04-23:
Here's a new MPC package in julia that supports C codegen
https://github.com/darnstrom/LinearMPC.jl
the docs are sparse and the level of testing isn't very high yet, but it could turn into something nice eventually if the author keeps working on it

franckgaga, 2025-04-23:
Yes It would be a simpler option than ACADOS, and it is generally expected that codegen only support a subset of the features. So if only DAQP solver is supported that would be okay IMO.

closed 2026-01-19 by franckgaga