JuliaControl/ModelPredictiveControl.jl #146: Error on execution of doc example

pekpuglia, 2025-01-10:
I tried to run [this example](https://juliacontrol.github.io/ModelPredictiveControl.jl/stable/manual/nonlinmpc/) and got a giant error at `sim!`, which begins with:

> ERROR: MethodError: no method matching ndims(::Type{PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}})
> 
> Closest candidates are:
>   ndims(::Base.Broadcast.Broadcasted)
>    @ Base broadcast.jl:247
>   ndims(::ArrayInterface.ArrayIndex{N}) where N
>    @ ArrayInterface ~/.julia/packages/ArrayInterface/a4Wwt/src/ArrayInterface.jl:851
>   ndims(::Base.Generator)
>    @ Base generator.jl:53
>   ...
> 
> Stacktrace:
>   [1] Base.Broadcast.BroadcastStyle(#unused#::Type{PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}})
>     @ Base.Broadcast ./broadcast.jl:103
>   [2] combine_styles(c::PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}})
>     @ Base.Broadcast ./broadcast.jl:448
>   [3] combine_styles(c1::PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, c2::Base.Broadcast.Broadcasted{Base.Broadcast.DefaultArrayStyle{1}, Nothing, typeof(identity), Tuple{Vector{Float64}}})
>     @ Base.Broadcast ./broadcast.jl:449
>   [4] materialize!
>     @ ./broadcast.jl:881 [inlined]
>   [5] (::ModelPredictiveControl.var"#inner_solver_f!#24"{DataType, RungeKutta, ModelPredictiveControl.var"#f!#35"{typeof(f)}, PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, PreallocationTools.DiffCache{Vector{Float64}, Vector{Float64}}, Float64})(xnext::Vector{Float64}, x::Vector{Float64}, u::Vector{Float64}, d::Vector{Float64}, p::Vector{Float64})
>     @ ModelPredictiveControl ~/.julia/packages/ModelPredictiveControl/KRRfW/src/model/solver.jl:56

The error appears to come from this line:

`@. xcur = x`

I'm using Julia 1.9.

franckgaga, 2025-01-10:
Thanks for the report. Probably because 1.9 is no longer officially support, the new LTS is 1.10.

franckgaga, 2025-01-10:
I just tried to run the motor exemple in the manual on Julia 1.9, and there is no error. Are you using the last stable version of the package (v.1.1.4) ?

franckgaga, 2025-01-11:
Weird I tested it today and now I can reproduce. I'll investigate on that.

franckgaga, 2025-01-14:
@pekpuglia, is the issue persisting with one of the two workarounds ?

pekpuglia, 2025-01-15:
Hello, thank you for your responses, I still haven't had the time to get back to this. Maybe by the end of the week, but this looks good. Will try migrating to julia 1.10.

pekpuglia, 2025-01-16:
Hi, switched to julia 1.10, works like a charm. Maybe add a [compat](https://pkgdocs.julialang.org/v1/compatibility/) line in Project.toml?

franckgaga, 2025-01-16:
Yes, I added the line in the last version of the package (v1.2.0).

closed 2025-01-16 by franckgaga