JuliaControl/ModelPredictiveControl.jl #163: Switch to `AbstractVector` and `AbstractMatrix` to support e.g. `StaticArrays.jl`

franckgaga, 2025-02-21:


franckgaga, 2025-02-22:
FYI @baggepinnen and @1-Bart-1

I experimented a bit with `StaticArrays.jl`. Here's some findings:

1. Converting everything to `SVector` and `SMatrix` significantly complexifies the code and reduce it readability a lot. It needs to be really justified IMO.
2. I would need to use mutating static arrays, since all the `struct` in this package are non-mutating (the default behavior of `struct`). Many methods in the package are currently allocation-free with builtin arrays because I use mutation a lot (ex: `updatestate!` mutate `.x0` or `.x̂0` vector, `setmodel!` and `linearize!` mutates `.A`, `.Bu`, `.Bd`, `.C` and `.D` matrices, etc.)
3. I ran the micro-benchmark script on my computer (Julia 1.11):
```
============================================
    Benchmarks for 3×3 Float64 matrices
============================================
Matrix multiplication               -> 2.7x speedup
Matrix multiplication (mutating)    -> 1.6x speedup
Matrix addition                     -> 16.6x speedup
Matrix addition (mutating)          -> 3.0x speedup
Matrix determinant                  -> 73.7x speedup
Matrix inverse                      -> 90.3x speedup
Matrix symmetric eigendecomposition -> 3.1x speedup
Matrix Cholesky decomposition       -> 13.8x speedup
Matrix LU decomposition             -> 4.2x speedup
Matrix QR decomposition             -> 31.6x speedup
```
the speedups of the mutating operations are modest. And it's micro-benchmarks: we cannot interpret these results as "the MPC controller would be 3.0x faster". These operations are generally not the bottleneck for MPC.

From these findings, I am not convinced at all this feature is worth the implementation.

1-Bart-1, 2025-02-23:
Using StaticArrays isn't very important to me, but the ability to use FiniteDiff would be great.

closed 2025-02-28 by franckgaga