JuliaControl/ModelPredictiveControl.jl #164: Direct collocation transcription

franckgaga, 2025-02-23:


franckgaga, 2025-04-05:
@1-Bart-1 still interested in this feature ?

1-Bart-1, 2025-04-06:
Thank you for asking! Difficult to say yet, improving the kite simulator took a couple of months more than expected 😆 
I will be experimenting with MPC.jl in the coming weeks, so I will come back to this.

franckgaga, 2025-04-06:
Alright I'll prioritize #177 in the next weeks for now, it's been a while that I want to implement MV blocking.

franckgaga, 2025-08-19:
Hello @baggepinnen! I'm working on the trapezoidal method and I'm stuck at a detail. You said this:

> The most efficient way to implement trapezoidal discretization tends to be to implement it as constraints to IPOPT. This way, you don't need any external root finder, Ipopt will handle everything through the constraints
> 
> $$F(x, u) = 0$$
> 
> This can also cut down the number of function evaluations by a factor of 2 (provided that you accept FOH instead of ZOH interpolations of the inputs), since you only need to evaluate the dynamics once per time step, whereas if you implement it as a time-stepping integrator with ZOH interpolation, you cannot exploit the fact that you have already evaluated $f(x_k, u_k)$ when you compute the step between $f(x_k, u_k)$ and $f(x_{k+1}, u_{k+1})$ when you computed the step $f(x_{k-1}, u_{k-1})$ to $f(x_k, u_k)$ 

 _Originally posted by @baggepinnen in [#140](https://github.com/JuliaControl/ModelPredictiveControl.jl/issues/140#issuecomment-2558426822)_

It feels weird to assume piece-wise constant (or FOH) manipulated inputs in the transcription method when the user will presumably implement the optimal solution on the real plant as piece-wise constant inputs (or ZOH). Or even in simulation, we can assume that the user will simulate the plant with a stepper like RK4, and they almost always assume piece-wise constant manipulated inputs. Two questions:

1. Do you think that this discrepancy between the transcription method and the reality will affect the closed-loop response? What's you experience on this?
2. If I would like to implement a trapezoidal method that assumes piece-wise constant manipulated inputs, to be a little bit more realistic, would it consist in computing the defects as :
```math
x_{k-1}-x_k-0.5\Delta T[ f(x_k, u_k) + f(x_{k+1}, u_k) ]
```
instead of:
```math
x_{k-1}-x_k+-0.5\Delta T[ f(x_k, u_k) + f(x_{k+1}, u_{k+1}) ]
```

franckgaga, 2025-08-20:
Yes, I meant computing the defects as:

```math
x_{k+1}-x_{k}-0.5\Delta T[ f(x_{k+1}, u_k) + f(x_k, u_k)  ]
```

instead of:

```math
x_{k+1}-x_k+-0.5\Delta T[f(x_{k+1}, u_{k+1}) +  f(x_k, u_k)]
```

Thanks!

P.S. The more I think of it, the more I find hidden complexities in collocation methods. In the package, I define my stochastic model for the unmeasured disturbances as discrete state-space. The $f$ function above should be the continuous state-space model of the plant model augmented with the unmeasured disturbance model. Do I need to call `d2c` on my stochastic model ? I'm not sure, but it feels like a dirty solution to me.