JuliaControl/ModelPredictiveControl.jl #181: Support exact Hessians of nonlinear objective and constraints in `NonLinMPC` and `MovingHorizonEstimator`

franckgaga, 2025-03-22:
@baggepinnen I think for Hessians, it would make sense to always default to sparse computations (when activated, I would keep the LBFGS approximation as the default)? Should it be pure ForwardDiff, or mixed mode?

franckgaga, 2025-03-22:
Also maybe @gdalle has tips on that subject ? Thanks!

gdalle, 2025-03-28:
If your problems are reaasonably small, pure-ForwardDiff sparse Hessians are not a bad idea.
Otherwise you'd have to combine backends, and that's where it gets tricky:

- ReverseDiff only remains alloc-free in the absence of constants
- Zygote is of course not alloc-free, nor does it support mutation
- Mooncake and Enzyme don't work as inner backends for ForwardDiff

Pure-Enzyme sparse Hessians would also be an option but at the moment they're not so optimized, and there are still bugs I don't understand with respect to `DI.Cache` in second-order Enzyme

franckgaga, 2025-04-05:
Alright let's always sparse forwarddiff by default. The cost might be prohibitive on large systems, be default setting will still be the LBFGS approximation anyway.

Any thoughts @baggepinnen 💭?

baggepinnen, 2025-04-05:
sparse ForwardDiff is probably the only "safe" default in the sense that it's most likely to work, all others have quite spotty support in my experience.

franckgaga, 2025-04-05:
Btw, how's your trip to Jordan ? 🇯🇴

Will visit the Iceland for the first time in July, really looking forward for this trip 🇮🇸

franckgaga, 2025-11-04:
Implemented in #275 and #277.

closed 2025-11-04 by franckgaga