JuliaControl/ModelPredictiveControl.jl #199: Use one function in linearization

1-Bart-1, 2025-05-04:
Combining the functions `f_x!, f_u!` and `f_d!` into one function `f_xud!` before linearization would reduce the number of function calls needed for linearization, and thus increase performance. This would give one big jacobian, which can be split up into the `A, Bu` and `Bd` matrixes.

https://github.com/JuliaControl/ModelPredictiveControl.jl/blob/9e800029610c43f27a9ad915d3ca21797abcbadf/src/model/linearization.jl#L33C1-L35C87

franckgaga, 2025-05-05:
I'm not convinced the increased code complexity and reduced clarity is worth the presumably small performance gain here. There is only two function calls to the state update function `solver_f!` if you don't have measured disturbances (i.e. `f_x!` and `f_u!`).

I'm not even sure that's possible, moreover. The `solver_f!` is differentiated with respect to a distinct variable on each `jacobian!` call i.e. with respect to $\mathbf{x}$ for $\mathbf{A}$, and with respect to $\mathbf{u}$ for $\mathbf{B_u}$:

[image]

The code right now is consistent with the classical linearization theory: one Jacobian computed for each state-space matrix.

1-Bart-1, 2025-05-05:
Ok, nevermind then!