JuliaControl/ModelPredictiveControl.jl #205: Support nonlinear estimator with linear model predictive control

1-Bart-1, 2025-05-20:
Is there any reason this cannot be done? It would improve the state estimation of non-measured states while still keeping good performance.

franckgaga, 2025-05-20:
It is possible right now only if the plant model is identical in the state estimator and in the linear mpc, hence only with linear plant model.

The only reason was to simplify the constructor (I fetch the plant model at `estim.model`, thus no need for an additional model argument when you specify a custom state estimator). Otherwise it could be done yes. 

My only concern is how to "enforce" that the state-space realization in the state estimator is close enough to the state-space realization of the controller model (if there is two arguments at construction). The only thing we can check is the state vector length, it seems to me.

Another solution would be #142. The `CustomEstimator` would include the necessary information about the linear plant model and the unmeasured disturbances (i.e. the augmentation scheme) to allow the construction of the linear MPC. An then calling `preparestate!` and `updatestate!` would do nothing except receiving an external $\mathbf{\hat{x}}$ value computed by the user.

@baggepinnen I think this feature is supported in JuliaSimControl ? How do you deal these aspects of that's the case?

1-Bart-1, 2025-05-20:
Both a `CustomEstimator` and a nonlinear estimator in `LinMPC` would be really useful to me.

franckgaga, 2025-05-20:
I suppose it would be for a successive linearization MPC? What I have in mind is an API like this:

```julia
model = NonLinModel( ... )
x_init, u_init = ...      # specify the initial linearization point here
linModel = linearize(model, x=x_init, u=u_init)
nint_ym = ...             # specify a custom model augmentation scheme here if needed
custom = CustomEstimator(linModel, nint_ym)
mpc = LinearMPC(custom)
estim = UnscentedKalmanFilter(model, nint_ym)
for i=1:10
    y = ...                             # get current measured outputs
    x̂ = preparestate!(estim, y)         # correct UKF state estimate
    preparestate!(mpc, y)               # do nothing at all, can be omitted
    setstate!(mpc, x̂)                   # update MPC with the UKF corrected state 
    u = moveinput!(mpc)
    x = x̂[1:end-sum(nint_ym)]
    linearize!(linModel, model, x, u)
    setmodel!(mpc, linModel)
    x̂ = updatestate!(estim, u, y)       # update UKF state estimate
    updatestate!(mpc, u, y)             # do nothing except store u as the lastu value
    setstate!(mpc, x̂)                   # update MPC with the UKF updated state
end
```
Would it be something that match your need ?

baggepinnen, 2025-05-21:
> I think this feature is supported in JuliaSimControl ? How do you deal these aspects if that's the case?

The optimizer isn't concerned with how the state estimate is produced, it can come from anywhere. In our higher-level simulation routines, we allow any estimator that produces a state of the correct dimension. In practice, it will of course only work if the state represents the same thing in the controller and optimizer model. The state of the "actual" system and the state of the estimator are separate things in JSC, making it easy to use any estimator with any model.

1-Bart-1, 2025-05-21:
> I suppose it would be for a successive linearization MPC?
Yes
> Would it be something that match your need ?
Yes I think that makes a lot of sense

baggepinnen, 2025-05-21:
NoEstimator? It sounds like the option turns off the built in estimator?

franckgaga, 2025-05-21:
Yes it turn off the builtin estimator. The object is only there to pass the necessary information to construct the predictive controller (e.g. the state augmentation scheme).

I like `ManualEstimator`, it says what it does. Go for it!

franckgaga, 2025-05-24:
FYI @1-Bart-1 

I decided to move the storing of `lastu` field in `moveinput!` instead of `updatestate!` (this value is needed to convert input increments to inputs). This change will slightly simplify the API for the upcoming `ManualEstimator`, that is, `updatestate!(mpc, u, y)` will be no longer needed:

```julia
model = NonLinModel( ... )
x_init, u_init = ...      # specify the initial linearization point here
linModel = linearize(model, x=x_init, u=u_init)
nint_ym = ...             # specify a custom model augmentation scheme here if needed
custom = CustomEstimator(linModel, nint_ym)
mpc = LinearMPC(custom)
estim = UnscentedKalmanFilter(model, nint_ym)
for i=1:10
    y = ...                             # get current measured outputs
    x̂ = preparestate!(estim, y)         # correct UKF state estimate
    preparestate!(mpc, y)               # do nothing at all, can be omitted
    setstate!(mpc, x̂)                   # update MPC with the UKF corrected state 
    u = moveinput!(mpc)
    x = x̂[1:end-sum(nint_ym)]
    linearize!(linModel, model, x, u)
    setmodel!(mpc, linModel)
    x̂ = updatestate!(estim, u, y)       # update UKF state estimate
    updatestate!(mpc, u, y)             # CHANGED: do nothing at all, can be omitted
    setstate!(mpc, x̂)                   # update MPC with the UKF updated state
end
```

closed 2025-05-24 by franckgaga