JuliaControl/ModelPredictiveControl.jl #238: New transcription method based on orthogonal collocation

franckgaga, 2025-08-30:
Similar as implemented here: https://www.do-mpc.com/en/latest/theory_orthogonal_collocation.html

franckgaga, 2026-02-10:
Hello @baggepinnen !

I'm starting to implement a new `OrthogonalCollocation` transcription method and I have a question. In the section [here](https://www.do-mpc.com/en/latest/theory_orthogonal_collocation.html#deriving-the-integration-equations), how do we compute the $a_{j,k}$ coefficients ? I took a look at Biegler (2010) and they provide the values for the polynomial roots (that is, the $\tau$ values), but not the time derivative of the Lagrange polynomial at the roots. What did you do in DyadControlSystems, you used `ForwardDiff.jl` for this ? edit: or hard-coded values bases on analytical derivatives ?

baggepinnen, 2026-02-11:
You can see my implementation here 
https://github.com/baggepinnen/SeeToDee.jl/blob/main/src/SeeToDee.jl#L452C2-L467C4