JuliaControl/ModelPredictiveControl.jl #257: Multithreaded `MultipleShooting` and `TrapezoidalCollocation`

franckgaga, 2025-09-17:
@baggepinnen 

I want to add conditional multi-threading for `MultipleShooting` and `TrapezoidalCollocation`, similar to DyadControl. I manage to get multi-threaded computation for the nonlinear equality constraints (the state defects over the prediction horizon) by using:

```julia
Threads.@threads for j=1:Hp
    # I fill the the equality constraint vectors `geq` here.
    # I never use Threads.threadid() and there is seperate
    # memory space for each iteration, so race conditions
    # are theoretically impossible.
end
```

Now what's the best way to do *condtionnal* multi-threading without code duplication ? The copilot suggested:

```julia
if use_threads
    Threads.@threads for j in 1:Hp
        # loop body
    end
else
    for j in 1:Hp
        # loop body
    end
end
```

but the loop body will be duplicated. I also tried to create this macro:

```julia
macro maybe_threads(use_threads, loop)
    if use_threads
        return esc(:(Threads.@threads $loop))
    else
        return esc(loop)
    end
end
```

but the only way to use it is with boolean literals, that is `@maybe_threads true for j=1:Hp` or `@maybe_threads false for j=1:Hp`.

Thanks!

baggepinnen, 2025-09-18:
The macro has to escape the `use_threads` variable if you want to use a non-literal

franckgaga, 2025-09-18:
Woah, I just understood that it's impossible to multi-thread the `TrapezoidalCollocation(1)`, with the FOH, using the tricks that calls the state derivative `f` function only once per time step. The iterations needs to be "re-orderable" for multithreading. If we call the `f` function only once per time steps, we need the results from the previous time steps, which may or may not be computed at the time of execution, giving a race condition.

Multi-threading can gets very tricky...

closed 2025-09-19 by franckgaga